ros2/examples #403: Question About Colcon Test

leizhenyu-lzy, 2025-01-18:
Environment : OS: Ubuntu 22.04.5 LTS x86_64
ROS Version: ros2/humble
Problem : Successfully run the `colcon build` or `colcon build --symlink-install`, but have problem in running `colcon test`

[image]

```bash

=============================== warnings summary ===============================
test/test_flake8.py::test_flake8
test/test_flake8.py::test_flake8
  Warning: SelectableGroups dict interface is deprecated. Use select.

-- Docs: https://docs.pytest.org/en/stable/warnings.html
---
Finished <<< launch_testing_examples [18.0s]

Summary: 22 packages finished [18.2s]
  10 packages had stderr output: examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher launch_testing_examples
```

What can I do to solve this problem?  Thank you.

leizhenyu-lzy, 2025-01-18:
Do I need to remove the folder "launch_testing_examples" ? Because in the documentation this folder do not exist.

[image]

fujitatomoya, 2025-01-21:
@leizhenyu-lzy can you provide the complete procedure from the beginning instead of posting the images?

> but have problem in running colcon test

if `colcon test` has any error, you can check the result with `colcon test-result --verbose` and the fix the failures.

> Do I need to remove the folder "launch_testing_examples" ? Because in the documentation this folder do not exist.

if you are following the examples, i do not think you need to remove them because those are the packages in [examples](https://github.com/ros2/examples). e.g https://github.com/ros2/examples/tree/rolling/rclpy/executors/examples_rclpy_executors

christophebedard, 2025-01-21:
"stderr output" doesn't mean that there were test failures: it simply means that some tests printed things to stderr, which could just be deprecation warnings.

If tests had failed, colcon would've printed `[ with test failures ]` next to `Finished <<< launch_testing_examples [18.0s]`.