ros2/examples #405: Examples on humble branch fails to compile on Docker ros/humble image

jhill515, 2025-02-28:
# Issue
Examples on humble branch fails to compile on Docker ros/humble image

# Steps to Reproduce
**Assumptions** - Default user is part of `docker` group

* On host machine, `docker run -it ros/humble bash`.
* In `ros/humble` image
  *  `mkdir -p ros_ws/src && cd ros_ws`
  * `git clone https://github.com/ros2/examples src/examples -b humble` (Note, sourced from https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html)
  * `colcon build`

# Resulting Errors
```
root@591a644c5f31:~/ros2_ws# colcon build
Starting >>> examples_rclcpp_async_client
Starting >>> examples_rclcpp_cbg_executor
Starting >>> examples_rclcpp_minimal_action_client
Starting >>> examples_rclcpp_minimal_action_server
Starting >>> examples_rclcpp_minimal_client
Starting >>> examples_rclcpp_minimal_composition
Starting >>> examples_rclcpp_minimal_publisher
Starting >>> examples_rclcpp_minimal_service
Starting >>> examples_rclcpp_minimal_subscriber
Starting >>> examples_rclcpp_minimal_timer
Starting >>> examples_rclcpp_multithreaded_executor
Starting >>> examples_rclcpp_wait_set
Starting >>> examples_rclpy_executors
Starting >>> examples_rclpy_guard_conditions
Starting >>> examples_rclpy_minimal_action_client
Starting >>> examples_rclpy_minimal_action_server
Starting >>> examples_rclpy_minimal_client
Starting >>> examples_rclpy_minimal_publisher
Starting >>> examples_rclpy_minimal_service
Starting >>> examples_rclpy_minimal_subscriber
Starting >>> examples_rclpy_pointcloud_publisher
Starting >>> launch_testing_examples
--- stderr: examples_rclcpp_minimal_action_server                                  
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findexample_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "example_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "example_interfaces" with any of the following names:

    example_interfacesConfig.cmake
    example_interfaces-config.cmake

  Add the installation prefix of "example_interfaces" to CMAKE_PREFIX_PATH or
  set "example_interfaces_DIR" to a directory containing one of the above
  files.  If "example_interfaces" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< examples_rclcpp_minimal_action_server [1.63s, exited with code 1]
--- stderr: examples_rclcpp_wait_set
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findexample_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "example_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "example_interfaces" with any of the following names:

    example_interfacesConfig.cmake
    example_interfaces-config.cmake

  Add the installation prefix of "example_interfaces" to CMAKE_PREFIX_PATH or
  set "example_interfaces_DIR" to a directory containing one of the above
  files.  If "example_interfaces" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< examples_rclcpp_wait_set [1.62s, exited with code 1]
--- stderr: examples_rclcpp_minimal_action_client
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findexample_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "example_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "example_interfaces" with any of the following names:

    example_interfacesConfig.cmake
    example_interfaces-config.cmake

  Add the installation prefix of "example_interfaces" to CMAKE_PREFIX_PATH or
  set "example_interfaces_DIR" to a directory containing one of the above
  files.  If "example_interfaces" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< examples_rclcpp_minimal_action_client [1.64s, exited with code 1]
--- stderr: examples_rclcpp_async_client
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findexample_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "example_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "example_interfaces" with any of the following names:

    example_interfacesConfig.cmake
    example_interfaces-config.cmake

  Add the installation prefix of "example_interfaces" to CMAKE_PREFIX_PATH or
  set "example_interfaces_DIR" to a directory containing one of the above
  files.  If "example_interfaces" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< examples_rclcpp_async_client [1.65s, exited with code 1]
--- stderr: examples_rclcpp_minimal_client
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findexample_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "example_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "example_interfaces" with any of the following names:

    example_interfacesConfig.cmake
    example_interfaces-config.cmake

  Add the installation prefix of "example_interfaces" to CMAKE_PREFIX_PATH or
  set "example_interfaces_DIR" to a directory containing one of the above
  files.  If "example_interfaces" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< examples_rclcpp_minimal_client [1.65s, exited with code 1]
--- stderr: examples_rclcpp_minimal_service
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findexample_interfaces.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "example_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "example_interfaces" with any of the following names:

    example_interfacesConfig.cmake
    example_interfaces-config.cmake

  Add the installation prefix of "example_interfaces" to CMAKE_PREFIX_PATH or
  set "example_interfaces_DIR" to a directory containing one of the above
  files.  If "example_interfaces" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< examples_rclcpp_minimal_service [1.65s, exited with code 1]
Aborted  <<< examples_rclpy_minimal_publisher [1.92s]                                             
Aborted  <<< examples_rclpy_guard_conditions [1.96s]
Aborted  <<< examples_rclpy_minimal_client [1.99s]                                                
Aborted  <<< examples_rclpy_executors [2.03s]
Aborted  <<< examples_rclpy_minimal_service [2.01s]
Aborted  <<< launch_testing_examples [2.06s]
Aborted  <<< examples_rclpy_minimal_subscriber [2.07s]                                             
Aborted  <<< examples_rclpy_pointcloud_publisher [2.15s]
Aborted  <<< examples_rclpy_minimal_action_client [2.17s]
Aborted  <<< examples_rclpy_minimal_action_server [2.17s]
Aborted  <<< examples_rclcpp_minimal_timer [7.05s]                                                
Aborted  <<< examples_rclcpp_minimal_publisher [9.54s]                                            
Aborted  <<< examples_rclcpp_multithreaded_executor [12.9s]                                         
Aborted  <<< examples_rclcpp_cbg_executor [13.4s]                                                    
Aborted  <<< examples_rclcpp_minimal_subscriber [14.2s]                                                    
Aborted  <<< examples_rclcpp_minimal_composition [15.3s]                                    

Summary: 0 packages finished [15.4s]
  6 packages failed: examples_rclcpp_async_client examples_rclcpp_minimal_action_client examples_rclcpp_minimal_action_server examples_rclcpp_minimal_client examples_rclcpp_minimal_service examples_rclcpp_wait_set
  16 packages aborted: examples_rclcpp_cbg_executor examples_rclcpp_minimal_composition examples_rclcpp_minimal_publisher examples_rclcpp_minimal_subscriber examples_rclcpp_minimal_timer examples_rclcpp_multithreaded_executor examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher launch_testing_examples
  6 packages had stderr output: examples_rclcpp_async_client examples_rclcpp_minimal_action_client examples_rclcpp_minimal_action_server examples_rclcpp_minimal_client examples_rclcpp_minimal_service examples_rclcpp_wait_set
```

# Expected Result
Clean-ish build? At least something that compiles with warnings.

christophebedard, 2025-02-28:
I believe it's because the `ros:humble` Docker image doesn't contain the `example_interfaces` package (`ros-humble-example-interfaces`). This is why the tutorial mentions this: https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#install-ros-2

[image]

The `ros:humble` image is not the "desktop" installation/image; it contains fewer packages. You can get that from the `osrf/ros:humble-desktop` image, see: https://hub.docker.com/_/ros.

jhill515, 2025-03-02:
I didn't realize `ros-humble-example-interfaces` was included in `ros:humble-desktop`. Since I was attempting to establish a devops pipeline, I tried to go without the extra support. For my use-case, I'm going to have to find the GitHub project for the interfaces so that I can demonstrate their build in conjunction with the example project.

christophebedard, 2025-03-02:
You could use `rosdep` to install only the dependencies of the packages in your workspace, see: https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html#rosdep-operation

jhill515, 2025-03-02:
Thanks! I'm going to close the issue -- Adding the `ros/example_interfaces` source resolved my issue.

closed 2025-03-02 by jhill515