ros2/examples #412: Non Clean Exit from Action Examples

Tanishq30052002, 2025-04-24:
### Operating System:

```Ubuntu 22.04```

### ROS version or commit hash:

rolling

### 'ros2 doctor --report' output

<details><summary>ros2 doc --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

kill server then kill client

if done in reverse order then server have a clean exit but client will have a non clean exit

### Expected behavior

Both should have a clean exit

### Actual behavior

[image]

fujitatomoya, 2025-04-26:
1st of all, https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debs.html#resources, ros2 rolling supports ubuntu noble. did you try to build the source on ubuntu jammy?

i tried with today's source build. this is what happens no matter what order we shutdown the application by SIGINT `Ctrl-C`.

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run examples_rclcpp_minimal_action_client action_client_member_functions
[INFO] [1745632529.548502487] [minimal_action_client]: Sending goal
[INFO] [1745632529.631025594] [minimal_action_client]: Goal accepted by server, waiting for result
[INFO] [1745632529.631231659] [minimal_action_client]: Next number in sequence received: 1
[INFO] [1745632530.631319485] [minimal_action_client]: Next number in sequence received: 2
^C[INFO] [1745632532.231458498] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1745632532.231532707] [rclcpp]: signal_handler(signum=2)
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to create guard condition: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/guard_condition.c:65
[ros2run]: Aborted

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
[INFO] [1745632529.630650768] [minimal_action_server]: Received goal request with order 10
[INFO] [1745632529.631014643] [minimal_action_server]: Executing goal
[INFO] [1745632529.631095635] [minimal_action_server]: Publish Feedback
[INFO] [1745632530.631228628] [minimal_action_server]: Publish Feedback
^C[INFO] [1745632531.308664879] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1745632531.308726399] [rclcpp]: signal_handler(signum=2)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Asked to publish result for goal that does not exist
[ros2run]: Aborted
```

this is expected exception since signal handler shuts down the context. of course we can obviously add `try...catch` statement to this exception, but what needs to be done after that? we can convert the exception into debug print, but i do not really think that is useful for user either. what could be useful for user as demonstration or example against this point is, having the application specific signal hander to manage the signal in the application layer to clean up the managed object, and the system signal handler shuts down the context? but i think that is also really application specific behavior.

Tanishq30052002, 2025-04-26:
I haven't checked on 24.04, but I remember the same kind of issues happens on ros2 humble apt installation in 22.04

Tanishq30052002, 2025-05-09:
@fujitatomoya any views over this?