ros2/examples #428: Add C++ single-goal action server example (preemption-like behavior)

araitaiga, 2025-09-28:
### Description

Add a C++ action server example that demonstrates single-goal behavior where new goals preempt existing ones. This would provide equivalent functionality to the existing Python `server_single_goal.py` example.

### Motivation

The Python examples include server_single_goal.py demonstrating single-goal behavior, but there is no C++ equivalent. This leaves C++ developers without a reference for goal preemption patterns commonly needed in robotics.

### Design / Implementation Considerations

- Add `rclcpp/actions/minimal_action_server/single_goal.cpp`
- Abort previous goal when new goal is accepted in `handle_accepted` callback

### Additional Information

- server_single_goal.py
  - https://github.com/ros2/examples/blob/rolling/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_single_goal.py

cc @fujitatomoya

fujitatomoya, 2025-09-28:
as mentioned in the issue report, this is just a counterpart of https://github.com/ros2/examples/blob/rolling/rclpy/actions/minimal_action_server/examples_rclpy_minimal_action_server/server_single_goal.py. i say this is nice to have as c++ example too.

@mjcarroll @christophebedard @ahcorde or anyone any objections? 🤔

araitaiga, 2025-10-07:
Thanks everyone for the feedback!  
I'll start working on the C++ example (`single_goal.cpp`) based on the Python version.  
Once it's implemented, I'll open a PR referencing this issue.